SKGleba/PSP2-batteryFixer #8: Unsupported battery controller rev.

leosel, 2020-09-08:
Is there's something wrong with my battery if this error displays on your app?

rand92000, 2020-12-07:
I would also like to know this, @SKGleba 
(Also - thank you for your work. :)

dothackjhe, 2021-05-05:
I also seem to be getting a similar issue on the PS Vita Slim.

RayCist1608, 2021-06-25:
I think it's an issue only on the slim

SKGleba, 2021-08-16:
Will be fixed in the next update

TootyFruityAnim, 2022-10-16:
yo when is the next update coming out? it's been a year

SKGleba, 2022-10-16:
Hopefully next week.

TootyFruityAnim, 2022-10-25:
bro its been 9 days and no update to vpk

Sheikhm007, 2023-01-02:
The app still shows an error in the vita slims, when will it be fixed ?

Brandon98SE, 2023-01-04:
Thanks for making this program, it just saved me money buying a new battery for my Red Japanese PS Vita my Mum got me for Christmas 2022! 👍 🥇

SKGleba, 2023-01-06:
fixed in v3.0

closed 2023-01-06 by SKGleba