SKGleba/PSP2-batteryFixer #13: PSVita doesn't restart

andiweli, 2023-01-07:
My PSVita Slim doesn't shut off. Instead the display goes grey and the Power LED blinks endlessly (I waited 5 minutes the first time). I have to manually press Power for 20 seconds to switch it off.

Afterwards I don't have to enter date and time.

YuuMorisawa, 2023-01-08:
I also had the same issue with my slim as well.

SKGleba, 2023-01-08:
Is anything shown on screen?
If yes, please tell me the battery controller info.

andiweli, 2023-01-08:
Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

SKGleba, 2023-01-08:
hmm 0xff01, different chip.
try the latest v3.0.1 release to make sure its not the race condition

andiweli, 2023-01-08:
Nope doesn't work.

Now I can see the text screen forever. After 1 minute I went back to the home screen and quit the app.
I restarted and it stays black/grey (with backlight)... long pressing the power button shows the canel/shutdown screen.
After rebooting I was shown the security boot screen, selected the first.
System started, no date to be set so far.

:(

mikesierra85, 2023-01-08:
> Nope doesn't work.
> 
> Now I can see the text screen forever. After 1 minute I went back to the home screen and quit the app. I restarted and it stays black/grey (with backlight)... long pressing the power button shows the canel/shutdown screen. Then the LED starts blinking again and it's in some kind of "loop".
> 
> Long pressing (10 seconds) the power button again reboots the Vita. After rebooting I was shown the security boot screen, selected the first. System started, no date to be set so far.
> 
> :(

I experienced the same behaviour as you, also 0xff01

James1o1o, 2023-01-10:
Same behaviour, same battery info

Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

TootyFruityAnim, 2023-01-15:
My PS Vita almost got bricked, I launched the vpk and it didn't restart, so I tried to restart the PS Vita myself and it got stuck on the loading screen, had to get it into safe mode and restart it from there and it didn't work.

MikalWhaley, 2023-02-07:
> Nope doesn't work.
>
> Now I can see the text screen forever. After 1 minute I went back to the home screen and quit the app. I restarted and it stays black/grey (with backlight)... long pressing the [...]

Yeah I get the same sort of thing now. Brand new slim I just unboxed the other day. Figured this would be the best tool if I can only charge to 90%. It's exactly the same as you mentioned though. First bootup only text no restart. Close the app try again, black screen. If I try again till get stuck loading the app and need to be hard reset.

carmelopaz0708, 2023-02-10:
I've gotten the same issue as others have mentioned in this thread. The app displays a static screen on my Vita 2000 and does not restart. Here are the details.

Console info:
Soc rev: 11.5
Cur firmware: 3.65
Min firmware: 3.65
PS TV emulation: Off
Secure state bit: 0x1
Manufacturing mode: No
Use QA (blank( PSID: No

Battery Controller info:
Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

dinckelman, 2023-02-27:
> I've gotten the same issue as others have mentioned in this thread. The app displays a static screen on my Vita 2000 and does not restart. Here are the details.

Identical system details to the previous comment. The app does not respond after opening. Pressing Start as indicated does not do anything. Makes the system unable to shutdown or open any apps naturally, requiring a hard restart. Upon restart, console goes into safemode, but lets you restart it again for a full boot

Galafador, 2023-06-21:
Console info:
Soc rev: 11.5
Cur firmware: 3.60
Min firmware: 3.50
PS TV emulation: Off
Secure state bit: 0x1
Manufacturing mode: No
Use QA (blank) PSID: No

Battery Controller info:
Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

Same issue to the previous comment. text stays forever, pressing start doesn't do anything. can't turn off the vita and have to enter Safe Mode afterwards.

oyashchyshyn, 2023-07-08:
I have same issue

Is there any updates about this?

fabiotph, 2023-09-02:
I have the same issue

Console info:
Soc rev: 4.2
Cur firmware: 3.65
Min firmware: 1.06
PS TV emulation: Off
Secure state bit: 0x1
Manufacturing mode: No
Use QA (blank) PSID: No

Battery Controller info:
Name: Abby
HW: 0xa3
FW: 0x413
DF: 0x4a
BOOT: 0xa3

Is there already a solution?

SKGleba, 2023-10-27:
A3 should work fine, could you try with an older release? pre-v3.

As for 0xff01, its a different abby version and has a slightly different transaction/handshake process, The problem is that syscon RE is painful due to the unfriendly assembly and state machine architecture. Will give it another shot next week, but more likely in december.

ru-el, 2023-11-29:
So I re-jailbreak my vita, tried the one from VitaDeploy which is outdated. It didn't worked (stuck in the Info screen) so I updated it from VitaDB and still didn't worked (it tried to restart but just got a blank screen with flashing Green LED).

Cosmycal, 2023-12-20:
Same problem here with my PSVita Slim with v3.0 goes to black screen with backlight on frozen there, and with v3.0.1 the text stays there and does nothing.

SoC rev: 11.5
Cur firmware: 3.65
Min firmware: 3.50
PS TV emulation: Off
Secure state bit: 0x1
Manufacturing mode: No
Use QA (blank) PSID: No

Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

Mawile0, 2024-01-31:
same issue here, tried several times with 3.0 and 3.0.1. 3.0 results in black screen with the LED blinking, 3.0.1 results in freeze on the text screen. both ways it crashes results in the vita needing to be rebooted into safe mode. as others have reported.

Soc: 11.5
Cur: 3.60
Min: 3.50
PSTV: Off
Secure state bit: 0x1
Manufacturing mode: No
Use QA PSID: No

Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

DracarisFire, 2024-04-22:
I have the same issue with the Vita 2000 as the other users have listed here. Crash/Infinite Blinking Green Light

Soc: 11.5
Cur: 3.60
Min: 3.50
PSTV: Off
Secure state bit: 0x1
Manufacturing mode: No
Use QA PSID: No

Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

rand92000, 2024-10-10:
> A3 should work fine, could you try with an older release? pre-v3.
>
> As for 0xff01, its a different abby version and has a slightly different transaction/handshake process, The problem is that syscon [...]

Since the project seems to be stuck and no one has responded to SKGleba's request to try pre-3.0
I confirm a temporary workaround for those still struggling (I have the Vita Slim [2000]): 
It works if ([v2.1 "Janus.1"](https://github.com/SKGleba/PSP2-batteryFixer/releases/tag/v2.1)) is installed
Newer versions v3.0, v3.0.1 didn't let me reach date and time setting.

ABerZz, 2025-01-01:
> > A3 should work fine, could you try with an older release? pre-v3.
> > As for 0xff01, its a different abby version and has a slightly different transaction/handshake process, The problem is that syscon RE is painful due to the unfriendly assembly and state machine architecture. Will give it another shot next week, but more likely in december.
> 
> Since the project seems to be stuck and no one has responded to SKGleba's request to try pre-3.0 I confirm a temporary workaround for those still struggling (I have the Vita Slim [2000]):
> 
> ### It seems to be working if ([v2.1 "Janus.1"](https://github.com/SKGleba/PSP2-batteryFixer/releases/tag/v2.1)) is installed
> Newer versions v3.0, v3.0.1 didn't let me reach date and time setting.

Previous versions (v2.1 and older) still may not work because of issue #8, which was fixed in v3.

My understanding (based on @SKGleba comment) is that v3 and earlier were meant to support _Abby_ chips, but revision 0xff01 of the chip is not correctly supported yet

segads, 2025-01-08:
Hello, the same happend to me, but I disconet the battery for about 5min and works but i wont charge, and if i  reboot it i display a no battery icon and the led just blink orange or stay orange if connected to the charger

ip-catbeep, 2025-04-26:
Hello,

I just wanted to add more details since this seems to affect Japanese Vita 2000 consoles for the most part. To my dismay, I've recreated this on two consoles already.

Running the default version on Vita Deploy resulted in a screen freeze that can only be fixed by a safe mode reboot.

Running Janus 2.1 results in issue [#8](https://github.com/SKGleba/PSP2-batteryFixer/issues/8) as stated by ABerZz.

Running 3.0 results in the black screen and a blinking green light.

Running 3.01 results in a screen freeze followed by the result that happens on 3.0 albeit with a little more delay.

One thing I do want to emphasize is that **the battery no longer charges whether it is on or off** ever since I ran the default version on Vita Deploy.

The console refuses to acknowledge any form of charge input while it's on, and when it's powered off the orange light indicator only stays on for less than a minute.

I've already removed the battery and performed the soft cycle. No luck so far on getting either of my consoles to charge. I haven't tried draining the battery flat so far.

Please place a warning to prevent other users from experiencing this issue. 

---

Soc: 11.5
Cur: 3.65
Min: 3.65
PSTV: Off
Secure state bit: 0x1
Manufacturing mode: No
Use QA PSID: No

Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01

CookiePLMonster, 2025-06-22:
> A3 should work fine, could you try with an older release? pre-v3.
>
> As for 0xff01, its a different abby version and has a slightly different transaction/handshake process, The problem is that syscon [...]

Did this ever goes anywhere? We likely just hit this issue on another Slim Vita with a battery revision now refuses to charge to over 30%, and we weren't aware of the compatibility issues until the console refused to reboot...

Cade-H, 2025-11-10:
Same issue here w/ 0xff01. Vita had not been powered on in 3+ years and now it will not charge past 88%.

Kramarbeiter, 2026-10-18:
Having the same issue.
Name: Abby
HW: 0xff01
FW: 0x11
DF: 0x7
BOOT: 0xff01
Japanese Verion of 2000.

Ran the program, had to force restart, now it wont load above 88%.

Will there ever be a fix? :/

Kramarbeiter, 2026-10-19:
So I found out something weird today, which first seemed like a fix for the 88% issue but then turned out wasn't one, I wanted to share it anyway. 

I tried to import quite a large amount of music (4+ GB) via VitaShell -> More -> Import. The process crashed during the import (due to energy saving mode I assume), but after I restated, my Vita showed 95% and counted down from there, while it wouldn't go beyond 88% before.

When I plugged in my charger, it even started to charge, but when I pulled it back out, it jumped back to the 88% and stayed there.

Maybe someone can do something with this piece of info. Happy to provide more details if needed.

Still hoping there will be a fix at some point.
Cheers.