SKGleba/PSP2-batteryFixer #14: Does not warn that charging cable must be disconnected

yllwboy, 2023-01-09:
The tool applies the fix correctly when disconnected but leaves the battery indicator in a bugged state if the charging cable prevents the tool from shutting down the system.

A warning in the dialog screen would be appreciated, a condition check would be even better.

innocentvoice, 2023-02-08:
i agree, i have the same issue.

SKGleba, 2023-10-27:
i did not know about that, and it should not be the case, i will look into it

kleftangel, 2023-11-30:
This helped me get it working. Thanks

CookerSfg, 2025-11-28:
Would have been nice to know. I did this and it fixed it. A warning before hand would have been useful

jfcwtf, 2026-08-31:
still no warning, tried to reopen after unplugging from pc and nothing. tried to power off, now its stuck spinning. wtf.