SKGleba/PSP2-batteryFixer #15: I think this bricked my vita...

DarkMageNugget, 2023-01-13:
I used this when my battery started acting up and now my ps vita won't boot up. It keeps thinking the battery is dead. Have charged it for a while, so that shouldn't be an issue.

Splash screen comes up, but then it immediately shows the dead battery symbol.

DarkMageNugget, 2023-01-13:
I stand corrected. It miraculously came back after 1 hour. After having this issue for an hour of it charging before turning it back on.

And it still isn't even close to charged. But it does turn on now. I guess it had to charge through the bug to finally start charging the battery.

Or it just hates me and wanted me to post before coming back. Either way, it is back and I can close this issue.

closed 2023-01-13 by DarkMageNugget

Faalangst26, 2023-09-05:
If anyone else runs into this issue, I had to leave it on the charger for multiple hours before it magically was able to be turned on again. No idea what caused it, using it on a 2000 model